ucb-bar/dsptools #9: Some tests fail now

grebe, 2016-11-09:
With the bump-firrtl branches of firrtl-interpreter and chisel-testers, I get these errors:

[info] Exception encountered when attempting to run a suite with class name: dsptools.numbers.BlackBoxFloatSpec *** ABORTED ***
[info]   java.lang.NoSuchMethodError: firrtl.FirrtlExecutionOptions$.apply$default$7()Lscala/collection/immutable/List;
[info]   at chisel3.BackendCompilationUtilities$$anon$1.<init>(Driver.scala:70)
[info]   at chisel3.BackendCompilationUtilities$class.compileFirrtlToVerilog(Driver.scala:68)
[info]   at chisel3.testers.TesterDriver$.compileFirrtlToVerilog(TesterDriver.scala:8)
[info]   at chisel3.testers.TesterDriver$.execute(TesterDriver.scala:61)
[info]   at chisel3.iotesters.ChiselRunners$class.runTester(ChiselSpec.scala:17)
[info]   at chisel3.iotesters.ChiselFlatSpec.runTester(ChiselSpec.scala:26)
[info]   at chisel3.iotesters.ChiselRunners$class.assertTesterPasses(ChiselSpec.scala:20)
[info]   at chisel3.iotesters.ChiselFlatSpec.assertTesterPasses(ChiselSpec.scala:26)
[info]   at dsptools.numbers.BlackBoxFloatSpec$$anonfun$6.apply$mcV$sp(BlackBoxFloat.scala:45)
[info]   at dsptools.numbers.BlackBoxFloatSpec$$anonfun$6.apply(BlackBoxFloat.scala:45)
[info]   ...

Similar for dsptools.numbers.DspComplexExamplesSpec

Does dsp-real-support-2 need to be merged? I'm confused- I thought we merged the blackbox stuff into master.

chick, 2016-11-11:
@grebe I will look at this tonight hopefully.  I do not think it is a merge problem with 
dsp-real-support.  The black box stuff has changed to actually use the stuff defined in dsptools.  It had been using a shadow implementation of real black boxes that lived in firrtl-interpreter

chick, 2016-11-11:
@grebe I'm pretty sure this is more version madness, All tests pass for me when I am on all main branches.  I think the problems was fixed with latest PR'S merged into chisel-testers

grebe, 2016-11-11:
Thanks, passes for me now too.

closed 2016-11-11 by grebe