ucb-bar/dsptools #15: examples.TransposedStreamFIRSpec fails

grebe, 2016-11-16:
I think this is because expect on negative SInts fails, so maybe it is a chisel-testers problem. I was running https://github.com/ucb-bar/chisel-testers/pull/61 - maybe this exposes the error. @chick, can you reproduce?

chick, 2016-11-17:
@grebe, I can reproduce it, working on a fix

chick, 2016-11-17:
@grebe this is very strange.  I can fix this by changing checkAnswer in ConstantTapTransposedStreamingTester to return a BigInt 
```
  def checkAnswer(n: Int) : BigInt = {
    // assumes inputs increase by 1 each time
    BigInt(c.taps.zipWithIndex.foldLeft(0) {case (s, (tap, idx)) => {
      s + tap * (if(n - idx >= smallest) n - idx else 0)
    }})
  }
```
I cannot reproduce the error in a simpler example.  I've been bitten by this before

chick, 2016-11-17:
I have spent even more time trying to simplify this problem to no avail.
In some specific and rare circumstances an implicit conversion from Int to BigInt fails when calling expect.  I have even tested whether having the caller and callee in different jars matters, it does not.

grebe, 2016-11-17:
That is very weird. The BigInt implicit conversion is a scala thing, right?

chick, 2016-11-18:
Yes, pure scala, except that other arguments to the function are chisel types. I can't believe that that is the cause though, and I did enough testing to suggest it's not some simple correlation.
BTW, I have submitted a PR to chisel-testers for this fix.

grebe, 2016-11-18:
Cool, thanks for tracking this down.

chick, 2017-01-11:
This was resolved, problem was based on wrong implicit being invoked

closed 2017-01-11 by chick