ucb-bar/dsptools #22: Should we add implicits to dsptools.numbers package object?

grebe, 2016-12-09:
I'm finding that 99% of the dsptools code I write requires the following:
```
import dsptools._
import dsptools.numbers._
import dsptools.numbers.implicits._
```

I think there is a use case for using dsptools without the numbers, but I'm not so sure it makes sense to use `dsptools.numbers` without the implicits.

chick, 2017-02-01:
@grebe is this still a decision that needs to be made

shunshou, 2017-02-01:
I think it should go into numbers. There's almost no reason to use numbers w/o implicits.

shunshou, 2017-02-01:
But if we're in agreement -- can I change it in my PR?

grebe, 2017-09-07:
We should revisit this

shunshou, 2017-09-07:
As long as typeclass added stuff is separate (doesn't affect expectations from base Chisel), you should just do it, I think.

PR #102 (2017-09-18): Importing dsptools.numbers._ now includes implicits by default
This exposed some interesting little weird things. I'll add some inline comments.

Closes #22
closed 2017-09-18 by grebe