ucb-bar/dsptools #35: Why did someone decide fractionalBits = binaryPoint?

shunshou, 2017-01-14:
binaryPoint is ambiguous. Could be # of bits from LSB or MSB before binaryPoint...

shunshou, 2017-01-14:
From a user standpoint, I also find it weird that FixedPoint is a chisel3 thing vs. DspReal is a dsptools thing. 

Also, one would think that DspReal and Real are similar, except Real is more like Ring vs. DspReal is more like FixedPoint, SInt, etc.

chick, 2017-01-17:
One of the motivating factors in the choice of binaryPoint was the desire to support binaryPoints that were negative indicating a virtual exponent.  I agree where the various players here (FixedPoint, DspReal) seems a bit arbitrary, but the short term decision was based on how fundamental FixedPoint needed to be in order to properly support it.  So it became part of chisel/firrtl, the real stuff was supported by the BlackBox/ExternalModule for chisel/firrtl and did not require any changes.  The utility of easily fungible Real/Fixed might argue to bring the real implementation down, but it does not seem to be on the short term critical path.

chick, 2017-01-17:
@shunshou did you have a specific proposal in this issue?  Change the name or simply better doc about the amibguities

shunshou, 2017-01-17:
Sorry, ok I forgot about the negative binary points. But yes, if that's fine, please document. I just have to think about that every time I generate FixedPoint (back translate from what I'm used to).

shunshou, 2017-02-01:
@chick I got confused again b/c "setBinaryPoint" = set # of fractional bits... Wasn't sure if it trimmed to the right # of fractional bits, shifted the binary point around, etc. etc. It's pretty confusing...