ucb-bar/dsptools #43: Set up Travis CI

grebe, 2017-01-16:
@chick @ucbjrl I don't have the permissions to set up travis. We should do this.

I'll add a .travis.yml if one of you don't mind setting up travis's side of things.

@ucbjrl is there a way to set it up to do nightly builds? I'd like to run a script that checks out the head of master of everything and check that it still works. Our last floating point regression shouldn't have been around for so long.

ucbjrl, 2017-01-16:
I'll look into what's involved in setting up travis.

We have Jenkins set up for chisel builds such that if anything in the
monitored group (chisel3, firrtl, chisel-testers, firrtl-interpreter,
rocketchip) changes, we rebuild everything. This eliminates most of the
motivation for nightly builds.

grebe, 2017-09-07:
Fixed by #99

closed 2017-09-07 by grebe