ucb-bar/dsptools #45: Mechanism to convert DspReal [that you know is >=0 b/c of the algorithm] to UInt?

shunshou, 2017-01-16:
I kind of need this to go from angles to discrete frequency bin # 's. I'm unsure about the best way to do this. 

At least for FixedPoint, you can theoretically do: FixedPoint -> Rount to SInt -> asUInt conversion...

DspReal you can round too... But is there some way to cast it as an Int and then convert it to an SInt? 

@chick @grebe 

I need something like this fairly urgently.

shunshou, 2017-01-16:
Ok, so someone can correct me: am I just looking for removeMatissa? (which is a function that's also misspelled...) @grebe @chick ?

grebe, 2017-01-17:
I think it is called .toUInt, but should probably be changed to asUInt

chick, 2017-01-17:
Sorry, but I'm not sure, I think @grebe is right.  It looks to me like it is just returning the underlying 64 bit real.  This seems asymmetric between FixedPoint and DspReal, does this currently work?

shunshou, 2017-01-17:
It's on my to do list of things to check. I actually want just the (signed) integer portion of the real..., possibly rounded.

shunshou, 2017-01-27:
@chick can you explain your warning?

```
// WARNING! May cause overflow!
11	module BBFToInt(
12	    input  [63:0] in,
13	    output reg [63:0] out
14	);
15	  always @* begin
16	  out <= $rtoi($bitstoreal(in));
17	  end
18	endmodule
```

grebe, 2017-01-27:
floating point numbers may be larger than maxint

shunshou, 2017-01-27:
oh ok makes sense lol

closed 2017-02-07 by shunshou