ucb-bar/dsptools #49: Do any of the features enabled by DSPContext actually work?

shunshou, 2017-01-17:
Or is it just a shell for yet-to-be implemented stuff?

I can live with overflow always growing, but it'd be good to, by default, register values for fixed multiply/add, etc. and it seems that's not done. Of course, if I go and change it directly in the type class, existing designs will be affected. 

Also, there's no option for 3 real multiplier complex multiplies? 

This matters more for me because I don't have a straight pipelined architecture, and there's a good amount of computation I need to break up into chunks...

grebe, 2017-01-17:
I don't think the fixedpoint typeclass looks at the dspcontext beyond picking between +& and +%.

I think the context should have an option for complex multiply style which the complex typeclass would consume.

shunshou, 2017-01-17:
Actually, this goes back to delay checking... but if you wanted to have registers after, say, every 2 muls, but not after every mul, then you wouldn't really be able to support it with DspContext, right? What would you suggest that I do? If I add support for registering, I could set default # multiplies to 0, so you guys don't have to change anything...

Anyways, I feel like I've been spending too much time sidestepping things and building on top of existing infrastructure rather than changing the infrastructure to do what it probably should've done... I'm just concerned with how things fundamentally affect the users {aka you for dsp, a bunch of other people from changes in base chisel/testers}.

grebe, 2017-01-17:
I think doing something that adds a register every other multiply is dangerous to do via some global-ish object like a context. I think it's kind of hard to do in chisel- I don't think you want to add a register every other time you call the multiply function, I think you want to have a register between after every pair of multiplies in your circuit. That's hard for chisel to grock and requires you write your code in a very particular way. I think that would make more sense as an annotation+firrtl pass.

Adding support for fixed number of registers after operations would be great, but the default should be 0.

shunshou, 2017-01-17:
Ok, then I guess this is lower priority and I'll only add if/when I need it. (I get this feeling adding registers after n multiplies via annotations would be more useful, but that's a whole different beast...... and maybe it's better to just add parameterized Regs directly in my modules for now)... unless @azidar has some opinion...

closed 2017-02-07 by shunshou