ucb-bar/dsptools #53: Hard "real" functions punched through the tester?

shunshou, 2017-01-20:
Would this be a possibility? For stuff like atan, sin, cos, or fixedtodouble... Could you have the inputs/outputs at whatever module layer be either punched through to top level or be peek/pokable so that you could take say peek an atan input, process it in scala, and then poke it to the result? That gives the whole golden model with swapping fixed/float at intermediate block levels that would be super cool to have directly in scala...

chick, 2017-01-20:
@shunshou this sounds pretty much what the scala implementation of the black boxes, BlackBoxFactory and allies, does. Take a look, maybe this is a documentation issue (I'll put this on dsp action items) or maybe a little bit of api work would make it better

shunshou, 2017-01-20:
@chick @grebe opinions for long term solutions?

shunshou, 2017-01-20:
Oops sorry didn't see. Wait is this a scala thing or a chisel thing? So you're saying then we don't need to rely on the fact that things aren't supported in verilator? Basically I still want to use verilator and pass it the correct results.

shunshou, 2017-01-20:
Any chance you'd be able to make a verilator compatible atan example? ;)

grebe, 2017-01-21:
I don't think anyone has done what you are describing with verilator. We have done that with the firrtl interpreter- the interpreter runs in the same jvm as the tester so it makes it easier to pass around blackbox models written in scala

chick, 2017-01-21:
@shunshou I think I just *actually* got what you are asking for here, and I don't think I know how to do this or whether it is practically possible.

shunshou, 2017-01-21:
Like theoretically, the bring out to top level thing should work (since, within the same clk cycle, you can peek and then poke based off of the peeked result). It's just a pain to bring out every custom function to top level that you'd then interact with via peek/poke...

shunshou, 2017-01-21:
Apparently you should be able to access internal signals in Verilator somehow? http://embecosm.com/appnotes/ean6/html/ch06s02.html
And same for VCS? I assume the PeekPokeTester would prevent you at least from poking internal signals for safety reasons?

grebe, 2017-09-07:
@shunshou would you say this issue is resolved? I think we found a working solution, although not as complete as what you originally envisioned.

shunshou, 2017-09-07:
I'm satisfied.

closed 2017-09-07 by grebe