ucb-bar/dsptools #75: Knowing when something is in a ring

chick, 2017-03-02:
The following circuit ignores the DspContext directive.
```
class BadUIntSubtractWithGrow extends Module {
  val io = IO(new Bundle {
    val a = Input(UInt(4.W))
    val b = Input(UInt(4.W))
    val o = Output(UInt(4.W))
  })
  val r = RegNext(DspContext.withOverflowType(Grow) { io.a - io.b })
  io.o := r
}
```
The reason is that since the IO's are not declared with generators that have been given a *Ring* constraint the - that is used is the basic chisel one and not the one in UInt Ring.  
As a result this circuit compiles, despite that subtract with overflow type *Grow* is not supported

chick, 2017-03-02:
@shunshou @grebe I don't know if this is a big problem but it tripped me up righting a test for unsupported subtraction

shunshou, 2017-03-02:
Actually -- that's confusing. If you have an import implicits, would it still not implicitly say that UInt is Ring? If you didn't import implicits, then I think that's correct... (not sure if I really understand implicits though...)

chick, 2017-03-02:
I did use
```
import dsptools.numbers.implicits._
```

shunshou, 2017-03-02:
ok... I'm not sure if I like that behavior then :\ Definitely would've been a gotcha for me when writing control logic (i.e. control always UInt, but I still want some of the custom behaviors enabled with the typeclasses).

chick, 2017-03-02:
I don't like it either, I couldn't figure out at first why I wasn't getting exception I expected :P but I'm not sure what to do about it.

shunshou, 2017-03-02:
Could you, with the implicit stuff do some type conversion from UInt to UIntTypeClass?

chick, 2017-03-09:
@grebe @shunshou 
I'm trying to test SignBit with the following
```scala
class SignBitTester(c: SignBit[UInt]) extends DspTester(c) {
  poke(c.io.uIn, 0)
  expect(c.io.uOut, 0.0)
}

class SignBit[TU <: Data : Ring](uGen: TU) extends Module {
  val io = IO(new Bundle {
    val uIn = Input(uGen)
    val uOut = Output(UInt(1.W))
  })

  io.uOut := io.uIn.signBit()
}
```
But the io.uIn.**signBit()** is a compile error.  What should I do to type signature to pick this up.

chick, 2017-03-09:
@shunshou also Numbers.md says the following, is that right
> a.signBit
> 1 if a is zero or positive; 0 if a is negative

shunshou, 2017-03-09:
signBit is only on Data:RealBits, Data:IntegerBits. Ring doesn't have any notion of "bits". RealBits and IntegerBits should be all encompassing (of Ring, Order, etc.). Sorry! Numbers.md is a typo -- it should be 0 if a is zero or positive; 1 is a is negative. Good catch!

chick, 2017-03-09:
I'll change the spec with my next commit of more tests

chick, 2017-03-09:
So what is the right way to apply signBit to parameter that could be UInt SInt, FixedPoint or DspReal

chick, 2017-03-09:
@shunshou Numbers section on div2 what does
```UInt: Consistent with a >> n (i.e. rounds 1/2 to 0)```
mean.  How can UInt be 1/2

shunshou, 2017-03-09:
```
class SignBit[TU <: Data : RealBits](uGen: TU) extends Module {
  val io = IO(new Bundle {
    val uIn = Input(uGen)
    val uOut = Output(UInt(1.W))
  })

  io.uOut := io.uIn.signBit()
}
```

Basically, Paul was using Data:Real to allow you access to anything more than +, -, * [Kind of confusing with DspReal] so I added some bit-related things by having RealBits extend Real.

shunshou, 2017-03-09:
I'm just saying that 1 >> 1 would return 0. Math-y 1 >> 1 should be 1 / 2^(1) == 1/2 (if you added more bit precision), but we don't do that.

shunshou, 2017-03-11:
@chick have you figured out how to give typeclass functionality to UInts? :\

like val test = UInt(3.W)

and then do some typeclass only thing on test? Should it just be done with some implicit conversion to UIntTypeClass? It's very confusing...

shunshou, 2017-03-12:
Actually, this is a huge problem that renders all of the typeclass stuff useless for all of my control logic.

chick, 2017-03-13:
I've fiddled with it a bit this weekend, I have not solved.  I'll work on it more tonight and tomorrow and hopefully by the end of hacking tomorrow, we can some up with something

chick, 2017-03-13:
I agree this is a big problem.

grebe, 2017-09-07:
I think this is resolved by our decision that operations should have their normal behavior (eg typeclass's + === UInt +) and the addition of context_+. I think this issue can be closed, although we should revisit how happy we are with the context_<operator> approach.

shunshou, 2017-09-07:
definitely dislike having context_, but cleanly separating out somehow is important.