ucb-bar/dsptools #88: BlackBoxFloat

shunshou, 2017-03-22:
@chick I get these warnings:

```
%Warning-COMBDLY: /Users/angie.wang/Chisel3FFAST/FFTGen/test_run_dir/FFASTTopTB/dspblocks.fft.FFASTTopSpec260459830/BlackBoxFloat.v:46: Delayed assignments (<=) in non-clocked (non flop or latch) block; suggest blocking assignments (=).
%Warning-COMBDLY: Use "/* verilator lint_off COMBDLY */" and lint_on around source to disable this message.
%Warning-COMBDLY: *** See the manual before disabling this,
%Warning-COMBDLY: else you may end up with different sim results.
%Warning-COMBDLY: /Users/angie.wang/Chisel3FFAST/FFTGen/test_run_dir/FFASTTopTB/dspblocks.fft.FFASTTopSpec260459830/BlackBoxFloat.v:26: Delayed assignments (<=) in non-clocked (non flop or latch) block; suggest blocking assignments (=).
%Warning-COMBDLY: /Users/angie.wang/Chisel3FFAST/FFTGen/test_run_dir/FFASTTopTB/dspblocks.fft.FFASTTopSpec260459830/BlackBoxFloat.v:172: Delayed assignments (<=) in non-clocked (non flop or latch) block; suggest blocking assignments (=).
%Warning-COMBDLY: /Users/angie.wang/Chisel3FFAST/FFTGen/test_run_dir/FFASTTopTB/dspblocks.fft.FFASTTopSpec260459830/BlackBoxFloat.v:16: Delayed assignments (<=) in non-clocked (non flop or latch) block; suggest blocking assignments (=).
```

Probably best to clean up?

chick, 2017-03-22:
@shunshou Shouldn't we be able to see these warning running tests in dsptools.  I tried changing all <= to =.  I didn't see error messages either way and tests passed just the same.  I tried running fft, but the git submodule update needed a password
```git@bwrcrepo.eecs.berkeley.edu's password:```
that I don't think I have

shunshou, 2017-03-23:
A couple of things: Seems like Verilator has a major bug, so I'm trying to switch over to VCS. VCS doesn't like the real blackbox stuff unless you add the -sverilog flag. I'll be making a PR at chisel-testers. I'm going to push the <= to = change to ring-refactor. Also, VCS didn't like out w/o reg for some ops. Still debugging and just feeling very sad and overwhelmed.