ucb-bar/dsptools #89: Complex + Mux

shunshou, 2017-03-24:
This shows up... not sure what's going on:

```
[info]   java.lang.IllegalArgumentException: requirement failed: can't create Mux with non-equivalent types dsptools.numbers.DspComplex@434 and dsptools.numbers.DspComplex@9e
[info]  
```

chick, 2017-03-24:
@shunshou That sounds like a recent chisel pr #450 that checks that both sides of a Mux have equivalent types. Do you have any idea what is triggering it in the source. 
@ducky do you think there might be a missed edge case with the new ```typeEquivalent```.

shunshou, 2017-03-24:
@chick So one side of the Mux is a Lit; another isn't. It's driving me absolutely crazy. I keep having to reassign constants to wires to get things to work. Other than that, the error message is 0 helpful.