ucb-bar/dsptools #90: Does fromDoubleWithFixedWidth work?

shunshou, 2017-03-24:
@chick See https://github.com/ucb-bar/chisel3/issues/558
I'm using fromDoubleWithFixedWidth, so if it's getting width = 26 instead of width = 28 for the constants, it implies this thing didn't work... I have no idea why... (unless Chisel also doesn't width pad in reality... despite supporting the construct). 

Can you please look into this for me? 

I'm just running into so many problems that I can't get any hardware done.

chick, 2017-03-24:
@shunshou I will look at this right now

chick, 2017-03-28:
@shunshou You were right, this was the source of the Borked problem.  I just got distracted by the OneHotMux stuff.  There may still be issues there but fixing this seems to have fixed the Borked example.  I have submitted a PR ucb-bar/chisel3#564.

shunshou, 2017-03-28:
@chick this is one source. Theoretically, I shouldn't have needed to specify width at all and Mux1H should've figured it out. I tried the WithFixedWidth thing in the hopes that it would fix the issue, but it didn't... hence the discovery that this was an issue.

Thanks for looking into this!