ucb-bar/dsptools #92: DspReal usage

shunshou, 2017-03-30:
Somewhat on the lines of what @stevobailey and @grebe did -- If we're using Verilator or VCS as the main mode of testing with Dsp-y things, it might make sense to have all DspReal nodes be 1 bit ports (since there's no notion of signed/unsigned there). The nodes should then be (via annotation or otherwise w/ the IfDef stuff) typed as "real" or not for simulation vs. synthesis. 

I don't think it's correct to do this on an AnalogType, because, really, AnalogType is more talking about directionality rather than how to represent the data. 

Opinions?

stevobailey, 2017-03-30:
For those playing the Verilog game at home, this makes DspReal basically the same as Verilog real. As before, it's not synthesizeable, but it is testable in VCS/NCSim.

Also, AnalogType has somewhat to do with directionality, but I think of it more as how the tools handle the signal. AnalogTypes should never be buffered or modified.

shunshou, 2017-04-02:
@chick Would this be possible to do? Add some kind of DspOption (annotation) that turns all DspReal's into 1 wire UInts (to Chisel), and in the Verilog emitter, (another option) output as in/out real port_name? 

We need 1 wire for synthesis and we need 1 wire + "real" for VCS simulation using 1 wire (rather than 64). In doing so, it'd also be necessary to feed in a different BlackBoxFloat that doesn't have all of the fromBits/toBits stuff (it should just stay in the real domain for non-synthesizable simulation). 

I need something like this ASAP so I can easily translate my Chisel tests to something that's actually compatible with tapeout...

chick, 2017-04-03:
@shunshou It sounds do-able, I'm looking into it.

chick, 2017-04-05:
@shunshou I am going to say yes, I think this is possible and I will work on it right away.  Can you give me any example code with some identifying comments on what things this new transform should be applied to.  An example of  the verilog output desired would be helpful too

grebe, 2017-04-05:
Perhaps the type in the DspReal blackboxes should just be Analog. Here is what we did to firrtl to make Analog work with `real` in simulation (warning: it's a hack). 

I added a var that controlled verilog emission for firrtl analog types [here](https://github.com/grebe/firrtl/commit/34bef6677c03131bf59474ba0fe406c42493c353). This should be done in a more clean way- maybe we should pull in @azidar here. I think the right solution is to replace Analog nodes with some working IR types that have control over what verilog they emit.

Then we added a transform+annotation to barstools [here](https://github.com/ucb-bar/barstools/commit/0816995ecbd5d2c3347ac627f07de144473d4455) (it probably belongs somewhere else). We define a mixin [here](https://github.com/ucb-art/craft2-chip/blob/c77f2adb76a1536470af0d22111a545172cedc26/src/main/scala/craft/RealAnalogAnnotator.scala#L22) that adds ``` `ifndef SYNTHESIS real `endif ``` to the declaration of anything analog.

This is all pretty hackish and I would really like to have a clean way to control verilog emission via annotations.

On a side note, is there a good function for barstools to call to look at annotations and load the appropriate transform classes? I'm sure it must exist in firrtl or chisel, I just don't know where.

shunshou, 2017-04-05:
I'm pretty sure, from our discussion last night, that you just need the line from the Driver that we looked at and it should work. I think most of the Generator stuff should be overhauled though... because Firrtl is a lot friendlier than when it was made.

chick, 2017-04-05:
@shunshou what line from the Driver, and which Generator are you referring to.

shunshou, 2017-04-06:
https://github.com/ucb-bar/chisel3/blob/master/src/main/scala/chisel3/Driver.scala#L151 (unrelated to DspReal)

chick, 2017-04-06:
@grebe @shunshou I have a branch of firrtl that ports your hack and cleans it up a little.  I don't think I quite see how the ```ifdef SYNTHESIS``` works, it looks to me like all that is replaced is the type name so If I do that substitution.  When I test with this I get
```
circuit Test :
  module Test :
    input a : Analog<16>
    output b : Analog<16>
    wire x : UInt<32>
```
becomes
```
module Test(
  input
`ifndef SYNTHESIS
  real
`endif
         [15:0] a,
  input
`ifndef SYNTHESIS
  real
`endif
         [15:0] b
);
  wire [31:0] x;
  assign x = 32'h0;
endmodule
```
I am probably missing something here.  I also am not that clear about whether this solves @shunshou 's urgent problem.  Please advise

shunshou, 2017-04-06:
It doesn't. I don't want to use Analog, because I can't use the chisel
testers to  simulate. I want to conditionally adapt DSPReal and modify the
testers to allow you to feed in real numbers, not their 64 bit
representations.