ucb-bar/dsptools #111: Can someone help me understand what's wrong?

shunshou, 2017-10-08:
https://github.com/ucb-bar/dsptools/blob/673888fedcbd3043773e1d5077d39a37fd2d4afc/src/test/scala/dsptools/VerboseDspTesterSpec/SimpleTBwGenTypeOption.scala#L162

I believe this test compiled, correct? 

Which would imply that something like:

`val test = -18.3.F(10.BP)` should also compile, right?

I'm trying to test ranges, and am running into problems with Chisel being incorrect. 

Consider: https://github.com/freechipsproject/chisel3/blob/ec827bef21263517140efd08c3b6dc31e309e52a/chiselFrontend/src/main/scala/chisel3/internal/firrtl/IR.scala#L95

Or more specifically:

```
case class SLit(n: BigInt, w: Width) extends LitArg(n, w) {
  def name: String = {
    val unsigned = if (n < 0) (BigInt(1) << width.get) + n else n
    s"asSInt(${ULit(unsigned, width).name})"
  }
  def minWidth: Int = 1 + n.bitLength
}

case class FPLit(n: BigInt, w: Width, binaryPoint: BinaryPoint) extends LitArg(n, w) {
  def name: String = {
    val unsigned = if (n < 0) (BigInt(1) << width.get) + n else n
    s"asFixedPoint(${SLit(unsigned, width).name}, ${binaryPoint.asInstanceOf[KnownBinaryPoint].value})"
  }
  def minWidth: Int = 1 + n.bitLength
}
```

LitArg has a width check to see if the width specified (w) is at least as large as minWidth (1 + n.bitLength). This is fine for FPList BUT `s"asFixedPoint(${SLit(unsigned, width).name}` creates an SLit that checks this again. Since n for SLit is the unsigned representation,  we know that minWidth should be n.bitLength (no +1). Unfortunately, currently width check always throws an exception when NEGATIVE FP or Interval lots are used. 

I've made a (temporary?) workaround in the ranges branch for Chisel [I can also make a PR to master, but I figure the only people using FP + Interval Lits are Paul and me right now..., so maybe it's OK to lump this in with whenever ranges become mainstream... and my workaround is probably something that others would find issue issue with], but this looks to very much be a bug. 

I just don't understand why it didn't throw the same error in the dsptools test. Is it because of the fact that I created a Lit in a Bundle??? 

Anyone have time to look into this? @chick @grebe 

Someone really needs to go through Range/FP related IR and stuff in Bits right now, 'cause I feel like it's half written. :o I had to make a lot of changes just to get any user code testable. :o And (as we were aware of before) it's also horribly out of sync with master branches, so patches would make this diverge even more from master... @_@ 

As an FYI -- you can keep track of which branches I'm using (excluding barstools, because I've removed anything referencing it for now) here (see update.bash): https://github.com/ucb-art/Chisel3DSPDependencies/tree/ranges

And the code I'm testing (for those w/ access) is at https://github.com/ucb-art/FFTGen
Specifically: https://github.com/ucb-art/FFTGen/blob/chisel3_transition/src/main/scala/dsptools/intervals/tests/AAPaperExample.scala

---

As a side note, @chick I can't use the branch of firrtl-interpreter that has the range analysis stuff with a combination of these other branches. Something was added in the combinational loop PR that messed things up (something about .copy??? not existing? -- I lost the actual compile message). 

Can you look into that? 

We can discuss more at our meeting, but I just wanted to get the issue out to seed the discussion and potentially any work before our meeting. 

---

Couple of other "I can't get proper math behavior" bugs out of range stuff before I check to see if ranges were inferred correctly... Hopefully, I can figure that out. It'd be nice if someone helped write out more simple test cases too, now that I have a way to hook up Interval IO and still use normal Chisel testers (which is what I spent the last couple of days on). 

Thx! :)

shunshou, 2017-10-09:
Update: On further inspection, it looks like the intervals branch diverged from master in some functionally incorrect way... :o at least wrt the implementation of FPLit.

ucbjrl, 2017-10-09:
I wonder if this is due to the change I made to get literals to work as
literals. Let's investigate tomorrow.

shunshou, 2017-10-09:
It wasn't. Paul and I investigated, and it looks like it comes from a change Chick made around ~Feb. when he first attempted to add in interval support. Master is correct; the intervals branch isn't.

chick, 2017-10-09:
I will look into this today.  I wonder if these branches are so stale we should first try to merge current masters, I'm not looking forward to that but it has to be done eventually anyway.  What do you think about that @azidar?

shunshou, 2017-10-09:
Would be good to figure out how long the merges will take, too. As an end user that wants to maybe get a paper out in the interim, the biggest help for me would be to probably just back-port some of the interpreter analysis stuff to the branch that I'm on (and I don't expect that to be too difficult to do?), but that's definitely not the "right" way to do things... haha... :(

closed 2017-10-28 by shunshou