ucb-bar/dsptools #112: PaperProject Details

chick, 2017-10-09:
Branches

| project | branch |
| ------- | ------- |
| firrtl | intervals |
| chisel3 | interval-type-2 |
| firrtl-interpreter | instrumenting-sizes |

shunshou, 2017-10-10:
Branches I'm using (based off the above):

Branches

| project | branch |
| ------- | ------- |
| firrtl | intervals-oct |
| chisel3 | interval-oct |
| firrtl-interpreter | ??? |

shunshou, 2017-10-11:
@chick can you make sure ops like the ones in https://github.com/ucb-bar/dsptools/tree/master/src/main/scala/dsptools/numbers are also supported? i.e. <<, >> not implemented:

scala.NotImplementedError: an implementation is missing
[info]   at scala.Predef$.$qmark$qmark$qmark(Predef.scala:225)
[info]   at chisel3.internal.firrtl.IntervalRange.$less$less(IR.scala:359)

chick, 2017-10-11:
Working on it.  Is there general agreement on what it means.  I would guess for shift that:
- Shifts the point of Closed and Open and BinaryPoint by the shift amount
- UnknownBounds and BinaryPoint remain unchanged.

And what about merge.  Not mentioned in [Wikipedia Intervals](en.wikipedia.org/wiki/Interval_(mathematics))
My instinct would be:
- increase the bounds of the interval with the smaller binary point so it matches the binary point of the larger
- min of lows
- max of uppers
- All this when defined. 
Or maybe more naively
- Low is min of bounds if defined, unknown otherwise
- Upper is max of bounds if defined, unknown otherwise
- max of binary point

@shunshou , @azidar @grebe Your thoughts?

shunshou, 2017-10-11:
So two things:

1) We need to look at how Adam's firrtl code does this -- probably would be useful to have Adam document user-facing features; or that might already be in his tests.

2) For dsptools:

a >> n
Arithmetic shift right where n is either Int or UInt
Note: precision loss will result since the decimal point location is not adjusted

AND

a.div2(n)
a/2^n
UInt: Consistent with a >> n (i.e. rounds 1/2 to 0)
SInt: Round output to nearest SInt via DspContext.trimType
FixedPoint: Adjusts decimal point; up to you how much precision to keep via DspContext.trimType and DspContext.binaryPointGrowth (adds more fractional bits on top of the input amount)
For both FixedPoint and SInt, again, be aware of overflow behavior when RoundHalfUp is used for the most positive input

^ Both should be eventually supported (but not necessarily in Chisel; can be in dsptools). 

What is merge?

It seems like, for example, with multiply, you were intending to have firrtl infer everything if uninferred i.e. maybe the naive approach?

Are we going at this expecting that Chisel basically should have no knowledge of intervals besides the ones that are user assigned and letting firrtl take care of pretty much everything?

shunshou, 2017-10-11:
Also, fyi, I pushed more changes to aforementioned branches to fix some bugs.

chick, 2017-10-11:
I added the unimplemented shift functions IntervalRange and Interval.  I need to add more tests but I wanted to get the implementations out there first.  I left merge() in as unimplemented.  Not sure what it is intended for.

chick, 2017-10-11:
And yes, I believe @azidar thought the naive approach of anything not simple with intervals should be deferred to Firrtl as best for now.

shunshou, 2017-10-11:
Ok then beyond that, the ops that should take precedence are clip, wrap, etc. (overflow behavior and truncation/round, etc.). After that, I think we're good for being able to try some more complicated legitimate hardware...

I will also add some tests...

shunshou, 2017-10-11:
Here are the ops Adam checked in IntervalMathSpec:

+
*
-
>, >=, <, <=, ===, =/=
>> int
<< int
<< uint
>> uint
[]()
arith assign?
cat

----

Is cat something built into the type in Chisel or an external helper? o_O

shunshou, 2017-10-11:
And as far as the rest of Adam's non-math tests, really it's:

Cannot defined unknown BP for interval (Chisel restriction) -- just my note

wrap 
clip

----

Is there a setBinaryPoint equivalent for Interval? That might be more necessary for stuff like rounding ops.

grebe, 2017-10-12:
I can't find a reference to merge in our doc. @chick, your instinct matches mine, but I'm still trying to think of why it would be useful.

chick, 2017-10-12:
@shunshou @azidar are there any explicit docs on clip, wrap, etc.  I don't see any implementation in firrtl on branch interval-oct, is there some other terms involved.

shunshou, 2017-10-12:
I don't think this is really the info you need, but in summary: https://docs.google.com/document/d/1UsaDir3rUp8CzKU4QHoPNd3XdBDh8tF99AyC0O02CJU/edit

Probably the info would be the kind that Adam will hopefully have time to write next week, but until he gets to that, your best bet would be to backtrack from the tests (which hopefully fully encapsulate what he wants):

https://github.com/freechipsproject/firrtl/blob/intervals-oct/src/test/scala/firrtlTests/interval/IntervalSpec.scala

chick, 2017-10-16:
@shunshou 
I've pushed naive wrap and clip to chisel3 intervals-oct, they don't really figure anything out right now. just make things Unknown so firrtl can figure it all out.  I need to talk to Adam to proceed on this.

Also I did not see a problem with running the interpreter's instrumenting-sizes branch with the intervals-oct branches of firrtl and chisel3, if you have an example can you pass it to me

shunshou, 2017-10-16:
Yeah, so Adam's super busy this week--he asked to reschedule our meeting (will send out an email shortly), but if he has a few minutes to meet with you tomorrow to discuss next steps, that'd be great. The one thing I'd like to see is being able to do clip based off of some other signal's range * # of loops, as an example, to handle: multiply by a constant and accumulate. I'll test firrtl-interpreter again with various branches and copy+paste the error I get in a few minutes.

shunshou, 2017-10-16:
Also, are you on the same branch of chisel-testers/dsptools as I am? 
Note: the last working verison of firrtl-interpreter is 5dd2f5b
When I switch over to firrtl-interpreter/instrumenting-sizes, and I try to re-publish-local chisel-testers with hash cb82340, I get:

```
[error] /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/src/main/scala/chisel3/iotesters/Driver.scala:130: type mismatch;
[error]  found   : optionsManager.type (with underlying type chisel3.iotesters.ReplOptionsManager)
[error]  required: firrtl_interpreter.InterpreterOptionsManager with firrtl_interpreter.HasReplConfig
[error]         FirrtlRepl.execute(optionsManager)
[error]                            ^
[error] /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/src/main/scala/chisel3/iotesters/Driver.scala:130: type mismatch;
[error]  found   : optionsManager.type (with underlying type chisel3.iotesters.ReplOptionsManager)
[error]  required: firrtl_interpreter.InterpreterOptionsManager with firrtl_interpreter.HasReplConfig
[error]         FirrtlRepl.execute(optionsManager)
[error]                            ^
[error] /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/src/main/scala/chisel3/iotesters/FirrtlTerpBackend.scala:14: type mismatch;
[error]  found   : chisel3.iotesters.TesterOptionsManager
[error]  required: firrtl_interpreter.HasInterpreterSuite
[error]   val interpretiveTester = new InterpretiveTester(firrtlIR, optionsManager)
[error]                                                             ^
[error] /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/src/main/scala/chisel3/iotesters/FirrtlTerpBackend.scala:14: type mismatch;
[error]  found   : chisel3.iotesters.TesterOptionsManager
[error]  required: firrtl_interpreter.HasInterpreterSuite
[error]   val interpretiveTester = new InterpretiveTester(firrtlIR, optionsManager)
```
If you are on a better version of chisel-testers (and corresponding dsptools), please let me know!

chick, 2017-10-17:
@azidar all the wrap cases I did made sense except for 
```scala
  val base = Wire(Interval(range"[-4, 6]"))
  val disjointRight = Wire(Interval(range"[7,11]"))
  val w6 = base.wrap(disjointRight)
  assert(w6 === 10.I())  // FAILS says it should be 6.U
                                       // even though derived interval is disjointRight Closed(7) Closed(11)
```
Any ideas

shunshou, 2017-10-17:
Before you delve too deep into this, as a basic use case, I just want to see that
val base = Wire(Interval(range"[0, 9]"))
val out = base.wrap(range"[0,7]")
base := 9.I 
--> out == 1.I without any mux logic. That's the most common use-case for wrap.

chick, 2017-10-17:
@shunshou I'll need to change chisel-testers to fix its publishLocal problems with the instrumenting-sizes branch of firrtl-interpreter.  Can there be a intervals-oct branch of chisel-testers that I can make my changes in. I'm not sure what to do with cb82340

shunshou, 2017-10-17:
yea go for it--make a branch and i'll point my stuff to it. thanks!! :D

chick, 2017-10-17:
@shunshou pushed my latest *wrap* stuff to intervals-oct.  This includes ```wrap(that: IntervalRange)``` which lets you say ```myInterval.wrap(range"[-4, 11)")``` if you are inclined to say such things
Will try to do clip next

chick, 2017-10-17:
@shunshou just pushed intervals-oct branch with interpreter fixes for instrumenting-sizes branch.
It's based on the cb82340 (I hope that's ok) this builds for me, hope it works for you

shunshou, 2017-10-17:
Hmm... I get 

```
] Executing in batch mode.
[warn]   For better performance, hit [ENTER] to switch to interactive mode, or
[warn]   consider launching sbt without any commands, or explicitly passing 'shell'
[info] Loading project definition from /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/project
[info] Set current project to Chisel.iotesters (in build file:/Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/)
[info] Packaging /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/target/scala-2.11/chisel-iotesters_2.11-1.2-SNAPSHOT-sources.jar ...
[info] Done packaging.
[info] Main Scala API documentation to /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/target/scala-2.11/api...
[info] Wrote /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/target/scala-2.11/chisel-iotesters_2.11-1.2-SNAPSHOT.pom
[info] :: delivering :: edu.berkeley.cs#chisel-iotesters_2.11;1.2-SNAPSHOT :: 1.2-SNAPSHOT :: integration :: Mon Oct 16 21:58:03 PDT 2017
[info] 	delivering ivy file to /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/target/scala-2.11/ivy-1.2-SNAPSHOT.xml
[info] Compiling 20 Scala sources to /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/target/scala-2.11/classes...
[error] /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/src/main/scala/chisel3/iotesters/ChiselPokeSpec.scala:15: type mismatch;
[error]  found   : options.type (with underlying type chisel3.iotesters.TesterOptionsManager)
[error]  required: chisel3.iotesters.TesterOptionsManager with firrtl_interpreter.HasInterpreterSuite
[error]     setupFirrtlTerpBackend(dutGen, options)
[error]                                    ^
[error] /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/src/main/scala/chisel3/iotesters/ChiselPokeSpec.scala:15: type mismatch;
[error]  found   : options.type (with underlying type chisel3.iotesters.TesterOptionsManager)
[error]  required: chisel3.iotesters.TesterOptionsManager with firrtl_interpreter.HasInterpreterSuite
[error]     setupFirrtlTerpBackend(dutGen, options)
[error]                                    ^
[error] /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/src/main/scala/chisel3/iotesters/Driver.scala:51: type mismatch;
[error]  found   : optionsManager.type (with underlying type chisel3.iotesters.TesterOptionsManager)
[error]  required: chisel3.iotesters.TesterOptionsManager with firrtl_interpreter.HasInterpreterSuite
[error]             setupFirrtlTerpBackend(dutGenerator, optionsManager)
[error]                                                  ^
[error] /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/src/main/scala/chisel3/iotesters/Driver.scala:60: type mismatch;
[error]  found   : Any
[error]  required: chisel3.iotesters.Backend
[error]         backendVar.withValue(Some(backend)) {
[error]                                   ^
[error] /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/src/main/scala/chisel3/iotesters/Driver.scala:62: type mismatch;
[error]  found   : dut.type (with underlying type Any)
[error]  required: T
[error]             testerGen(dut).finish
[error]                       ^
[error] /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/src/main/scala/chisel3/iotesters/Driver.scala:130: type mismatch;
[error]  found   : optionsManager.type (with underlying type chisel3.iotesters.ReplOptionsManager)
[error]  required: firrtl_interpreter.InterpreterOptionsManager with firrtl_interpreter.HasReplConfig
[error]         FirrtlRepl.execute(optionsManager)
[error]                            ^
[error] /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/src/main/scala/chisel3/iotesters/Driver.scala:51: type mismatch;
[error]  found   : optionsManager.type (with underlying type chisel3.iotesters.TesterOptionsManager)
[error]  required: chisel3.iotesters.TesterOptionsManager with firrtl_interpreter.HasInterpreterSuite
[error]             setupFirrtlTerpBackend(dutGenerator, optionsManager)
[error]                                                  ^
[error] /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/src/main/scala/chisel3/iotesters/Driver.scala:60: type mismatch;
[error]  found   : Any
[error]  required: chisel3.iotesters.Backend
[error]         backendVar.withValue(Some(backend)) {
[error]                                   ^
[error] /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/src/main/scala/chisel3/iotesters/Driver.scala:62: type mismatch;
[error]  found   : dut.type (with underlying type Any)
[error]  required: T
[error]             testerGen(dut).finish
[error]                       ^
[error] /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/src/main/scala/chisel3/iotesters/Driver.scala:130: type mismatch;
[error]  found   : optionsManager.type (with underlying type chisel3.iotesters.ReplOptionsManager)
[error]  required: firrtl_interpreter.InterpreterOptionsManager with firrtl_interpreter.HasReplConfig
[error]         FirrtlRepl.execute(optionsManager)
[error]                            ^
[error] /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/src/main/scala/chisel3/iotesters/FirrtlTerpBackend.scala:12: type mismatch;
[error]  found   : chisel3.iotesters.TesterOptionsManager
[... cut ...]
[error]  required: chisel3.iotesters.TesterOptionsManager with firrtl_interpreter.HasInterpreterSuite
[error]     optionsManager: TesterOptionsManager with HasInterpreterSuite = new TesterOptionsManager)
[error]                                                                     ^
[error] /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/src/main/scala/chisel3/iotesters/FirrtlTerpBackend.scala:106: type mismatch;
[error]  found   : chisel3.iotesters.TesterOptionsManager
[error]  required: chisel3.iotesters.TesterOptionsManager with firrtl_interpreter.HasInterpreterSuite
```

chick, 2017-10-17:
@shunshou just pushed clip (saturate) operator for intervals
BTW Aaaargh, on weird errors with testers, I'll try to reproduce

shunshou, 2017-10-17:
Thx! Finally have a ~10 pg outline of the ~6 pg paper. :P Will look into this after I'm done drafing background + intro hopefully...

chick, 2017-10-17:
Hmnm, I am building with
(cd firrtl ; git status | grep 'On branch')
On branch intervals-oct
(cd firrtl-interpreter ; git status | grep 'On branch')
On branch instrumenting-sizes
(cd chisel3 ; git status | grep 'On branch')
On branch intervals-oct
(cd chisel-testers ; git status | grep 'On branch')
On branch intervals-oct
(cd dsptools ; git status | grep 'On branch')
On branch master

shunshou, 2017-10-18:
@chick, I'm still getting:

```
error] /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/src/main/scala/chisel3/iotesters/ChiselPokeSpec.scala:15: type mismatch;
[error]  found   : options.type (with underlying type chisel3.iotesters.TesterOptionsManager)
[error]  required: chisel3.iotesters.TesterOptionsManager with firrtl_interpreter.HasInterpreterSuite
[error]     setupFirrtlTerpBackend(dutGen, options)
[error]                                    ^
[error] /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/src/main/scala/chisel3/iotesters/ChiselPokeSpec.scala:15: type mismatch;
[error]  found   : options.type (with underlying type chisel3.iotesters.TesterOptionsManager)
[error]  required: chisel3.iotesters.TesterOptionsManager with firrtl_interpreter.HasInterpreterSuite
[error]     setupFirrtlTerpBackend(dutGen, options)
[error]                                    ^
[error] /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/src/main/scala/chisel3/iotesters/Driver.scala:51: type mismatch;
[error]  found   : optionsManager.type (with underlying type chisel3.iotesters.TesterOptionsManager)
[error]  required: chisel3.iotesters.TesterOptionsManager with firrtl_interpreter.HasInterpreterSuite
[error]             setupFirrtlTerpBackend(dutGenerator, optionsManager)
[error]                                                  ^
[error] /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/src/main/scala/chisel3/iotesters/Driver.scala:60: type mismatch;
[error]  found   : Any
[error]  required: chisel3.iotesters.Backend
[error]         backendVar.withValue(Some(backend)) {
[error]                                   ^
[error] /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/src/main/scala/chisel3/iotesters/Driver.scala:62: type mismatch;
[error]  found   : dut.type (with underlying type Any)
[error]  required: T
[error]             testerGen(dut).finish
[error]                       ^
[error] /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/src/main/scala/chisel3/iotesters/Driver.scala:130: type mismatch;
[error]  found   : optionsManager.type (with underlying type chisel3.iotesters.ReplOptionsManager)
[error]  required: firrtl_interpreter.InterpreterOptionsManager with firrtl_interpreter.HasReplConfig
[error]         FirrtlRepl.execute(optionsManager)
[error]                            ^
[error] /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/src/main/scala/chisel3/iotesters/Driver.scala:51: type mismatch;
[error]  found   : optionsManager.type (with underlying type chisel3.iotesters.TesterOptionsManager)
[error]  required: chisel3.iotesters.TesterOptionsManager with firrtl_interpreter.HasInterpreterSuite
[error]             setupFirrtlTerpBackend(dutGenerator, optionsManager)
[error]                                                  ^
[error] /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/src/main/scala/chisel3/iotesters/Driver.scala:60: type mismatch;
[error]  found   : Any
[error]  required: chisel3.iotesters.Backend
[error]         backendVar.withValue(Some(backend)) {
[error]                                   ^
[error] /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/src/main/scala/chisel3/iotesters/Driver.scala:62: type mismatch;
[error]  found   : dut.type (with underlying type Any)
[error]  required: T
[error]             testerGen(dut).finish
[error]                       ^
[error] /Users/angie.wang/Chisel3FFAST/Chisel3DSPDependencies/chisel-testers/src/main/scala/chisel3/iotesters/Driver.scala:130: type mismatch;
[error]  found   : optionsManager.type (with underlying type chisel3.iotesters.ReplOptionsManager)
[error]  required: firrtl_interpreter.InterpreterOptionsManager with firrtl_interpreter.HasReplConfig
[error]         FirrtlRepl.execute(optionsManager)
```

with the branches you've pointed me to :(

chick, 2017-10-18:
Still trying to get this
Maybe now (had to push to interpreter and testers)
firrtl
3abda10e5e82b481f58c73adf26c48678e496574
firrtl-interpreter
4f0c623ad03b1b7f5c484aa99575ea324d8cdfbd
chisel3
14cc99a8880ba2ee735f888cec70c7cc99235a4b
chisel-testers
496bd86063b9a47a0bb13264068efdf94bd98a66

chick, 2017-10-18:
@shunshou What's the branch of dsptools

shunshou, 2017-10-18:
I assumed you were on master, so I switched to that (but that's probably just not compatible w/ old stuff...).

shunshou, 2017-10-18:
Follow up @chick @grebe 

https://github.com/ucb-art/Chisel3DSPDependencies/tree/ranges (range branch!) points to stuff that should hopefully work now:

firrtl: intervals-oct
chisel3: intervals-oct
firrtl-interpreter: instrumenting-sizes
chisel-testers: intervals-oct
dsptools: intervals-oct

shunshou, 2017-10-19:
@chick bpsl, bpsr, bpset should also be implemented in chisel, if it hasn't already.

chick, 2017-10-19:
I have just pushed to intervals-oct new methods on Interval
.setBinaryPoint(that: Int)
.shiftLeftBinaryPoint(that: Int)
.shiftRightBinaryPoint(that: Int)
I am not sure they are implemented exactly right, I need to check with Adam, but you can start to try them

shunshou, 2017-10-19:
Cool. I'm still in the process of debugging some of the shifts... Adam and I will be having a working meeting today ~1 to maybe up to 3ish? We should just clean out these details and make basic tests for everything while people are here... :o

chick, 2017-10-20:
I have started a documentation page on chisel3 wiki for [Intervals](/freechipsproject/chisel3/wiki/Interval-Type-in-Chisel-Experimental)

shunshou, 2017-10-20:
Ok, I'm still debugging some intermediate shr weirdness -- doesn't seem like I'm getting the right answer when I'm shifting by less than the width either :\, but need to finish up slides for another conference tonight. Thanks for starting the intervals documentation :D

shunshou, 2017-10-21:
@chick what implications are there if chisel range inference is wrong? (and firrtl is right?)

chick, 2017-10-22:
It would imply to me that things are broken.  In general, @azidar and I have tried to match the firrtl behavior, and in many cases actually use the firrtl code generate answers when we can.  I think there a number of cases where chisel might defer to firrtl, by just leaving things *unknown* so firrtl will figure it out, but it shouldn't contradict it

shunshou, 2017-10-22:
OK. I went ahead and double checked stuff and made things unknown wherever I felt there was a contradiction... Still not quite at the point of passing enough of my own tests, but getting closer.

shunshou, 2017-10-22:
@chick -- looks like x.clip(uint/sint) and x.wrap(uint/sint) don't work. Namely, there's no firrtl PrimOp that operates on 1 Interval and 1 UInt/SInt type... I thought I could handle this at the Chisel level by first doing asInterval on the UInt -- when the width is known, everything's good... unfortunately, when the width is unknown, I don't think there's any mechanism to pass the unknown (?) through from the Chisel emitter... [at least for asInterval?].

chick, 2017-10-22:
Hmm, it looks to me like ```x.wrap(sint)``` might be a problem, but the
rest of the family convert the argument to an interval before passing down to firrtl.
But clearly I need to expand the tests to include use of the SInt and UInt versions of 
clip and wrap, also to add tests for all these with some arguments with unknown ranges.  Sorry, but it's going to be later tonight before I can get to it.

shunshou, 2017-10-22:
Hm ok I made some changes assuming known UInt and SInt. I'm making my own
tests for them locally (basically assuming my model is intended behavior).
If you can generalize to handle unknown ranges, that'd be great. Also,
looks like some of my changes broke some of your tests. I made some of them
based off of my understanding of how functions should operate (from various
discussions), so we should make sure our understandings of behaviors match
and better document.

shunshou, 2017-10-23:
FYI--when adding tests, it's important to test stuff that's not just integers. Running into annoying binary point alignment problems, etc.

shunshou, 2017-10-23:
@chick to save me some time of trying to figure this out -- can you make a toy example for how firrtl interpreter stuff works with the driver in DspTools? (Likely stuff will need modification?)

chick, 2017-10-24:
Almost got it working, but found some edge case problems :-(.

shunshou, 2017-10-24:
no worries. i'm still trying to get accumulate working. :\ slow process...

shunshou, 2017-10-24:
Re: Interpreter -- 

I think we agreed on trimming widths based on some n sigma criteria or min/max. The other thing that's useful is that, if we're trimming based off of sigma, it should be trim and optionally clip to maximum.

chick, 2017-10-25:
The toy example works with my latest push/merges to 
firrtl-interpreter and dsptools on intervals-oct branches
See InstrumentingSpec for how.
Time to start working on feedback system

shunshou, 2017-10-25:
Yay! OK. I will try to run the interpreter on my interval arithmetic testbench and get back to you.

shunshou, 2017-10-28:
@azidar What do you expect should happen in these cases?

1) Directly assign a <= b where a has known range that's smaller than b. Should that be an error without explicit b.wrap(a), b.clip(a), or b.assignInterval(a)? Also, re: back-propagation of ranges, if b has some smaller bound and b <= x - y, I assume you can't make any assumptions about the ranges of x and y. However, if you've constrained x and y to be only positive and use only multiply and add ops i.e. b <= x + y, you can further optimize x and y ranges -- should be min(x_range, b_range). That's not something the tool should be able to figure out, maybe? But it'd be nice to have some optional Firrtl pass go and trim widths that way, maybe (unless synthesis tools are smart about that). 

2) Directly assign a <= b where a's bp = 5 and b's bp = 12 (both known). You're losing precision, which means that your interval also changes. In that case, before range inference, it might make sense to explicitly add b.bpset(a)? -- Otherwise, you're propagating the wrong ranges, which potentially accumulates and becomes probelamtic.

shunshou, 2017-10-28:
@chick @azidar @grebe fyi firrtl is on intervals-oct2 now.

shunshou, 2017-10-29:
@chick @grebe -- Intervals type classes should hopefully be in a workable state, so can actually *really* start using them. QOR for the firrtl output looks like what I calculated + the firrtl-interpreter sims got the right mins + maxes for the one test I paid closest attention to. I'm going to port over my testing infrastructure from my local repo to dsptools so hopefully it'll be easier for you to play with stuff. Should be up in the next hour or so? Then I'll go and try more complicated circuits and run them through some real synthesis tool.