ucb-bar/dsptools #139: DspContext with extra pipeline stages won't work for FixedPoint

tampler, 2018-09-29:
Is that an expected behaviour ? 

```scala
class MyMod[A <: Data:Ring, B <: Data:Ring] (inType:A, outType:B, mulPipeLen:Int = 3, addPipeLen:Int = 3) extends Module {
  val io = IO (new Bundle {  
    val a   = Input(inType.cloneType)
    val b   = Input(inType.cloneType)
  })
  val mres  = Reg(outType.cloneType) // mul result
  val ares  = Reg(outType.cloneType) // add result
 
  // Pipeline regs
  val a_q   = RegEnable(io.a, io.ena)
  val b_q   = RegEnable(io.b, io.ena)
...
    DspContext.withNumMulPipes(mulPipeLen) {
      mres := a_q * b_q
    }  
    DspContext.withNumAddPipes(addPipeLen) {     
      when (true.B) {
        ares := acc + mres
      }
```

Both mres and ares get translated into combinational logic in Verilog, altho the number of extra wires is equal to the num of pipeline stages provided.

tampler, 2018-09-30:
Okay, I found an answer to my question in one of examples. To make things working as expected, one needs to use **context** operations inside the **DspContext** block.
```scala
 DspContext.withNumMulPipes(mulPipeLen) {
      mres := a_q context_* b_q
    }  
    DspContext.withNumAddPipes(addPipeLen) {     
      when (true.B) {
        ares := acc context_+ mres
      }
```
Otherwise, it infers a default number of stages, which is 0

grebe, 2018-09-30:
Yeah, this is unfortunate. The issue is that it can be really confusing if `a + b` is using the typeclasses or not, so we decided that the vanilla operators (`+`, `*`, `-`, etc.) should not incorporate the context stuff and instead made a separate set of operators prefixed by `context_`. It's not the most satisfying solution, but I think it's better to be clear. I'd be curious to hear your thoughts.

I've wondered if having funky operators (e.g. `+$`, `-$`, `*$`, etc., or something similar), but I think this is potentially even more confusing.

tampler, 2018-10-01:
I thinks this solution is okay, given the wide scope of applications and the variety of data types those Ring-based operators are expected to work with. 

The next move is to improve documentation and code examples, since those are very basic operations and the use case with your library.

Could I contribute in this? I'm using your library to build my stuff and have enough insight in its intrinsics to help other develops to join the ride

shunshou, 2018-10-01:
Feel free to contribute with pull requests. The bulk of the documentation
is at
https://github.com/ucb-bar/dsptools/blob/master/src/main/scala/dsptools/numbers/README.md
(probably
should be moved to the top) but it looks like the examples I included were
either from before we decided on the context switch or I had a brain fart
and forgot to use the context version.... I’m also interested in what other
kinds of features or changes that you as a user want since only the handful
of us doing DSP at Berkeley have ever “tested” this as far as I know.

tampler, 2018-10-01:
Can I synch with your team? You definitely have high priority tasks, which you need to resolve. I see you're struggling with v1.1.4 release and you can't resolve dependencies in trunk.

As for PRs, I personally not interested in branching the lib itself. I'd prefer to work with your team. If you are not interested, I'll join another Berkeley project

shunshou, 2018-10-01:
As a rule, I don’t believe any Berkeley projects allow you to push directly
to master, regardless of whether you’re working with Berkeley on dev or
helping to patch things as an end user—we always use the PR/code review
process where we merge our local branch after the code has been code
reviewed. We’ve been looser about dsptools because there have only been a
handful of users (primarily ourselves, and right smack in the middle of
tapeout no less). There’s actually a pretty large PR from my side that’s
waiting to be merged once some PRs from Firrtl and Chisel are merged into
master. Ideally the PR would be from a branch with a manageable amount of
code to review/one particular fix or new feature. You can check out Chisel
or Firrtl PRs as an example. If it looks like things were directly pushed
to master recently, it’s because a large change from a different repo was
merged in; or rather, rebased in... but it still originated from a PR.

tampler, 2018-10-01:
Okay, I'm building a prop system and use your lib for FixedPoint and Complex computation. I had own Verilog implementation in past, but now switched to Chisel and deployed your lib. 

I can do PR with doc/tests/features/updates/examples. What should I do then? I still need a discussion. I can't upload the prop code, but I can PR high priority features, which we could agree to work on. 
I'm thinking about joining FIR, Rocket and Boom projects as well, but currently spend most time on DSP, so your project seems to be the best one to start with.

tampler, 2018-10-01:
Sure, what's the mail ? I have an idea. I need a stochastic rounding for my project, which is not supported within your lib. I'll start from a PR for this

shunshou, 2018-10-01:
You can e-mail me angie.wang@berkeley.edu and I can forward it out. I've
received some inputs on how we should go about doing rounding (mostly where
rounding should be implemented -- either at the output of operations or on
assignment), and I'd be curious what your preference is. We have some
simple rounding modes in our typeclasses, but additional flavors would
definitely be very welcome.

closed 2018-10-07 by tampler