ucb-bar/dsptools #149: Code Compilation fail in scala 2.12.8

tampler, 2018-12-17:
Project failed to compile in a new scala. Code needs to be refactored.
sbt 1.2.7
scala 2.12.8

> runMain examples.SimpleComplexAdder

[info] Compiling 42 Scala sources to /home/bku/work/bar/dsptools/target/scala-2.12/classes ...
[error] /home/bku/work/bar/dsptools/src/main/scala/dsptools/tester/DspTester.scala:229:5: overloaded method value expect with alternatives:
[error]   (data: dsptools.numbers.DspComplex[_],expected: breeze.math.Complex)Boolean <and>
[error]   (data: chisel3.Data,expected: Double)Boolean <and>
[error]   (signal: chisel3.experimental.FixedPoint,expected: Double)Boolean <and>
[error]   (signal: chisel3.experimental.FixedPoint,expected: Int)Boolean <and>
[error]   (signal: chisel3.SInt,expected: Int)Boolean <and>
[error]   (signal: chisel3.UInt,expected: Int)Boolean <and>
[error]   (signal: chisel3.Bool,expected: Boolean)Boolean <and>
[error]   (signal: chisel3.Bits,expected: BigInt,msg: => String)Boolean <and>
[error]   (good: Boolean,msg: => String)Boolean <and>
[error]   (signal: chisel3.Bundle,expected: Map[String,BigInt])Boolean <and>
[error]   (signal: chisel3.core.Aggregate,expected: IndexedSeq[BigInt])Boolean <and>
[error]   [T <: chisel3.core.Element](signal: T, expected: BigInt, msg: => String)(implicit evidence$14: chisel3.iotesters.Pokeable[T])Boolean
[error]  cannot be applied to (chisel3.Bits, scala.math.BigInt)
[error]     expect(signal.asInstanceOf[Bits], if (expected) BigInt(1) else BigInt(0))

grebe, 2018-12-19:
This will need to be done again once a snapshot is published with the new `Pokeable` API.

grebe, 2018-12-19:
Actually, it looks like my workstation was using an old snapshot and a snapshot with `Pokeable` has been published. The PR was updated.

PR #150 (2018-12-24): DspTester needs overrides now.
Closes #149.
commit 058e1243 (2018-12-24): DspTester needs overrides now. (#150)

* DspTester needs overrides now.

Closes #149.
closed 2018-12-24 by grebe