ucb-bar/dsptools #175: Issue with TileLink ready-valid logic in TLMasterModel

tymcauley, 2019-10-21:
In the [TLMasterModel Trait](https://github.com/ucb-bar/dsptools/blob/2b6c009456914c3a4aa86b1a6bd7b870b128abfd/rocket/src/main/scala/tl/TestIP.scala#L50), there are [`tlWrite*`](https://github.com/ucb-bar/dsptools/blob/2b6c009456914c3a4aa86b1a6bd7b870b128abfd/rocket/src/main/scala/tl/TestIP.scala#L76) and [`tlRead*`](https://github.com/ucb-bar/dsptools/blob/2b6c009456914c3a4aa86b1a6bd7b870b128abfd/rocket/src/main/scala/tl/TestIP.scala#L107) functions for each TileLink channel. All of them contain logic that looks something like this:

```scala
def tlWriteA(a: AChannel): Unit = {
  poke(memTL.a.valid, 1)
  pokeA(a)

  while(peek(memTL.a.ready) != BigInt(0)) {
    step(1)
  }
  step(1)
  poke(memTL.a.valid, 0)
}
```

I'm a bit confused by the line `while(peek(memTL.a.ready) != BigInt(0)) {`. That would suggest the logic of this function looks something like this:

* Assert `valid`.
* Wait for `ready` to **de**assert.
* Wait a cycle, then deassert `valid`.

Since the TileLink transaction fires when both `ready` and `valid` are high, shouldn't we wait for `ready` to assert, not deassert? Should that line read `while(peek(memTL.a.ready) == BigInt(0)) {` or `while(peek(memTL.a.ready) != BigInt(1)) {`?

grebe, 2019-11-06:
Sorry for the slow reply.

Yeah, you're absolutely correct. `TLMasterModel` has always been a bit rickety compared to AXI, perhaps this is one reason why. I'll assign fixing this to myself. The model should really be tested more.

tymcauley, 2019-11-06:
No problem, thanks for the confirmation! Let me know if I can do anything to help out. I can certainly submit a PR to fix these logical bugs in the `TLMasterModel` trait, but since I'm not so familiar with the code base, I don't know if I'll really be able to add new tests (which it sounds like you'd like to do).